Open-Acidification/TankController #5: Add ramp function for pH and Temperature setpoints

KirtOnthank, 2020-04-23:


prestoncarman, 2021-02-20:
@all-contributors please add @KirtOnthank for bugs, code, and ideas

jgfoster, 2021-08-24:
Two set-points and a time for each (before, during, and after).

jgfoster, 2021-08-25:
When asking for a new pH or temperature, there will be a follow-up question of the number of minutes to allow to get to the new value. The default minutes will be zero to provide backwards compatibility. If a non-zero value is entered, then the current time and target will be the starting point to the ramp and the newly-entered target and the current time plus the number of minutes will be the ending point. Each target query (typically to control the solenoid, but also to report to the screen, serial, and log) will be calculated based on a linear ramp between the start time/target and the end time/target. This will require storing three new values: start time/target and end time. The existing target will be associated with the end time.

eucalvo, 2021-10-31:
I believe this is the formula:
currentTarget = startingValue + ((currentTime  - startingTime) * (newTarget - startingValue) / (endTime - startTime))

eucalvo, 2022-03-01:
I believe these issues have been addressed by #266 and #267. Let me know if you see any issues.

closed 2022-03-01 by eucalvo