Open-Acidification/TankController #64: Reconsider size of v2 data in EEPROM

eucalvo, 2021-03-25:
const int AMPLITUDE_ADDRESS = MAC_ADDRESS + 8;
const int FREQUENCY_ADDRESS = AMPLITUDE_ADDRESS + 8;
const int GRANULARITY_ADDRESS = FREQUENCY_ADDRESS + 8;     // granularity for SD logging interval
const int MAX_DATA_AGE_ADDRESS = GRANULARITY_ADDRESS + 8;  // max data age for SD card
const int PH_SERIES_SIZE_ADDRESS = MAX_DATA_AGE_ADDRESS + 8;
const int PH_SERIES_POINTER_ADDRESS = PH_SERIES_SIZE_ADDRESS + 8;
const int TEMP_SERIES_SIZE_ADDRESS = PH_SERIES_POINTER_ADDRESS + 8;
const int TEMP_SERIES_POINTER_ADDRESS = TEMP_SERIES_SIZE_ADDRESS + 8;
const int PH_INTERVAL_ADDRESS = TEMP_SERIES_POINTER_ADDRESS + 8;
const int PH_DELAY_ADDRESS = PH_INTERVAL_ADDRESS + 8;
const int TEMP_INTERVAL_ADDRESS = PH_DELAY_ADDRESS + 8;
const int TEMP_DELAY_ADDRESS = TEMP_INTERVAL_ADDRESS + 8;

copied from https://github.com/Open-Acidification/TankControllerLib/issues/93

prestoncarman, 2021-06-10:
@all-contributors please add @Paydenb for bug

prestoncarman, 2021-06-10:
@all-contributors please add @eucalvo for bug

jgfoster, 2021-06-29:
Knowing that a `double` is 4 bytes, we can close this.

closed 2021-06-29 by jgfoster