Open-Acidification/TankController #78: Investigate licenses for included libraries

jgfoster, 2021-04-12:


prestoncarman, 2021-05-25:
Current List of Libraries and Licences

* Arduino-CI -  Apache-2.0 License
* Adafruit_BusIO -  MIT License
* Adafruit_MAX31865 - BSD license
* Ethernet - GNU Lesser General Public License
* LiquidCrystal - LGPL-2.1 License
* RTClib - MIT license
* Keypad - GPL-3.0 License
* Arduino-PID-Library - MIT License
* SD - GPL-3.0 License

jgfoster, 2021-06-10:
https://codebender.cc/library/Keypad#Keypad.cpp is LGPL 2.1
https://github.com/greiman?tab=repositories has some SD libraries that are MIT licensed

jgfoster, 2021-07-01:
It turns out that the Keypad library files have an LGPL 2.1 license and with #212 we have switched to a more recent library.

PR #234 (2021-08-06): Identify libraries and licenses
Fixes #78.
commit c4f413d7 (2021-08-06): Identify libraries and licenses (#234)

* Fix #78.

* Spelling

* Add link to libraries.md from README.md.
Also document way to avoid long DHCP request.

* Spacing for Codacy?

* More spacing and fixing a newline.
closed 2021-08-06 by jgfoster