Open-Acidification/TankController #132: Include Git SHA in version

jgfoster, 2021-05-05:
`echo "#define GIT_HASH " `git rev-parse --short HEAD` > GitHash.h`

prestoncarman, 2021-05-18:
Stack overflow has a few options: 
<https://stackoverflow.com/questions/1704907/how-can-i-get-my-c-code-to-automatically-print-out-its-git-version-hash>

jgfoster, 2021-09-05:
The challenge here is that for the value to in the program it must be in a .cpp or .h file. But since we want to reflect the current Git state, any edits to source code will make the Git checkout dirty. One solution is to regenerate the file on each compile, and the Arduino IDE has hooks for this, but you need to edit the toolchain (but see [this](https://github.com/arduino/Arduino/issues/11663)).

closed 2023-07-19 by jgfoster