Open-Acidification/TankController #275: Print actual tests that fail to stdout in testAndBuild.sh

pianistrevor, 2022-02-04:
Right now, test failures can be viewed in output.txt, but the command-line summary only includes how many failures are there. It would be nice to see which tests fail, for convenience; this also makes me ask how specific we can (or should) go: should sub-test failures be included as well?

pianistrevor, 2022-02-04:
Note: might also investigate including serial output in either output.txt or stdout.

jgfoster, 2022-02-06:
I hadn't noticed that `output.txt` had something so different from standard output. I tried the following:

```
bundle exec arduino_ci.rb --min-free-space=5960 1> >(tee stdout.txt ) 2> >(tee stderr.txt >&2 )
```

This shows that some of the output is going to stderr. I think it would be helpful to put errors to stderr and success to stdout.

pianistrevor, 2022-02-23:
I've added a PR to Arduino_CI that will output most things to stdout (with appropriate flush statements). It also does some formatting fixes.

pianistrevor, 2022-04-13:
Retiring this due to discussion with Ian regarding segfaults and the need to use unit buffering.

closed 2022-04-13 by pianistrevor