Open-Acidification/TankController #330: Handle GET for favicon.ico

jgfoster, 2022-08-19:
Get a small octopus icon and hardcode it into the program.

je-foster, 2022-08-26:
It appears that favicons have gotten more complicated in the 20 years since I last looked at them. There appear to be various sizes/formats, including very large ones for Android and iOS. [Here is a zip file based on the OPEN\nACID\nPROJ square graphic.](https://github.com/Open-Acidification/TankController/files/9434410/favicon_io.zip) The files total 138 KB. We currently have about 156 KB of program storage space available. Here are my ideas:

- We could include only the old-school "favicon.ico" (15 KB), sending 404's for the other requested files.
- We could redirect these requests to a different server, adding to the CORS/mixed content issues.
- We could load the 138 KB of files onto every SD card. Perhaps this could be automated through TankControllerManager.
- Is it possible to intercept these requests with TankControllerManager? Some of the website/scripts are running on localhost or cs.wallawalla.edu, right?

jgfoster, 2022-09-08:
> Is it possible to intercept these requests with TankControllerManager?

Actually, we could put an HTML element in the header to point in a different direction. See [this](https://stackoverflow.com/questions/10218178/necessary-to-add-link-tag-for-favicon-ico).

jgfoster, 2023-03-10:
This is 48x48 and 15,406 bytes.
[favicon.ico.zip](https://github.com/Open-Acidification/TankController/files/10937663/favicon.ico.zip)

jgfoster, 2023-07-11:
See https://stackoverflow.com/questions/8707183/how-can-i-convert-a-binary-file-to-the-text-declaring-a-c-c-array-with-that-co and the attached.
[favicon.c.txt](https://github.com/Open-Acidification/TankController/files/12021286/favicon.c.txt)

commit 91124c98 (2023-07-13): Provide a favicon.ico (#405)

Fix #330.
closed 2023-07-13 by jgfoster