Open-Acidification/TankController #361: Mobile first design

prestoncarman, 2023-03-06:
The primary use cases are through a mobile device.

With the exception of:

* importing time series using csv
* exporting data in csv format

jgfoster, 2024-01-09:
I think this is mostly making sure that the appropriate functionality works on a small screen. That is, we can still have a web app (no need to build for Android and iOS yet).

lydiathiel, 2024-01-12:
The current version of the web app seems to work fine for mobile use. There are other changes that could be made to the app to make it more user friendly- like refining the layout of the tank navigation menu, or adding highlighting on hover for file links- but the functionality is currently just as accessible to mobile devices as on a bigger screen. I've included screenshots from my iPhone as examples. 
[image]
[image]
[image]

jgfoster, 2024-01-12:
I'm going to close the issue for now. It can be reopened (or replaced) if we want more mobile features.

closed 2024-01-12 by jgfoster