Open-Acidification/TankController #383: Add functionality for PUT requests

Suocesky, 2023-04-10:
Add functions to accept and handle PUT requests in addition to POST requests so that values can be edited from the web app.

jgfoster, 2023-04-12:
The goal of this task is to support editing selected fields from the current information table (see EthernetServer_TC.cpp:132). The general format will be similar to a POST but is a PUT with a key from the list in JSONBuilder.cpp:37-49 and a new value for that field. On success the server should return a status code of 200 and the body content from the `current()` function (above).

jgfoster, 2023-04-17:
The steps we will do for this include the following:

- Fork to your personal GitHub account
- Clone to a local environment or to Code.CS in a Ruby workspace (our application is in C++ but the testing framework is Ruby)
- Copy the tests for POST but make it for PUT, confirm that they fail, then add/edit code to them pass
- Change the tests to support different keys, confirm that they fail, then make them pass
- Change the tests to support different return values (`current()` instead of `display()`), confirm that they fail, then make them pass

jgfoster, 2023-06-12:
Fixed by #389.

closed 2023-06-12 by jgfoster