Open-Acidification/TankController #403: Query slope on startup and after pH calibration

jgfoster, 2023-07-11:
- Warn if pH slope is out of range (initially hard-coded at 95-105%).
- Show on LCD with rapid blinking pH

jgfoster, 2023-07-28:
* We will add two new boolean flags to the system:
  * A "bad calibration" boolean with the probe instance (volatile); and
  * An "ignore bad calibration" boolean in the EEPROM (nonvolatile).
* On a pH calibration, we will _synchronously_ clear the "ignore bad calibration" flag.
* On startup and after a pH calibration, we will send an _asynchronous_ slope request to the device. 
* On receipt of a slope response, we will save the string and set the "bad calibration" flag based on the response.
* The main menu loop will rapidly blink if "bad calibration" is true.
* The main menu loop will check both flags and if "bad calibration" is true and "ignore bad calibration" is false, then the device will switch to a "bad calibration" UI state (which is "in calibration" so disable the bubbler) and present a screen that asks the user whether to ignore the bad calibration (which sets the nonvolatile flag) or start a new calibration.

je-foster, 2023-08-01:
I think that the EEPROM default value is 0xffffffff which represents the float -1 which converts to the boolean true. So instead of an "ignore bad calibration" boolean in EEPROM, I suggest a "warn for bad calibration" boolean. Then the default behavior is to warn about an out-of-range calibration.

je-foster, 2023-08-01:
This issue is resolved by #427.

closed 2023-08-01 by je-foster