Open-Acidification/TankController #434: Cancel pH calibration

je-foster, 2023-08-21:
### Current behavior
After the user performs a calibration of the pH probe, the Tank Controller checks the new slopes. If at least one of the slopes is not between 95% and 105% of ideal, then the Tank Controller warns the user that this appears to be a bad calibration. The user has two options:

1. Override the warning, accepting the new calibration
2. Reset to the probe's default ideal calibration

### Desired behavior
Replace option 2 so that the user has two options:

1. Override the warning, accepting the new calibration
2. Cancel the new calibration, restoring the previous calibration

je-foster, 2023-08-21:
To implement this, we would need to use the Export command for the EZO probe before starting the new calibration. We would need to save the calibration strings. If the user later chooses to cancel the new calibration, we would need to use the Import command to send the calibration strings back to the probe.

The calibration strings could be stored on the SD card, but I think that storing in a local variable would be satisfactory.

je-foster, 2023-08-21:
This could be implemented using #435 by "copying" to server before calibration begins and "pasting" from server in the event of a canceled calibration. However this would not be ideal, because Tank Controllers can be configured to not use a cloud server, or the network may not be available when a calibration is performed.

jgfoster, 2024-01-09:
I believe the documentation for EZO probe is [here](https://files.atlas-scientific.com/pH_EZO_Datasheet.pdf) and pages 23 and 24 describe the export and import process. From the documentation it appears that there are 120 bytes of data (hexadecimal digits) in an export, which should comfortably fit in RAM (and could be placed in EEPROM if we are worried about loss of data in event of a power interruption).