Open-Acidification/TankController #435: Export/Import pH calibration

je-foster, 2023-08-21:
### Scenario
Sometimes in the lab it is necessary to replace a defective pH probe on one Tank Controller with a working pH probe taken from another Tank Controller.

Although the pH probe may already have been calibrated on the second Tank Controller, it must be recalibrated after connecting to the first Tank Controller. Calibrations are specific to each physical pH probe, but they are saved on the Tank Controllers.

### Desired Feature
Transfer the calibration from the second Tank Controller to the first Tank Controller so that a new calibration does not need to be performed. One way to do achieve this would be with the following two steps:

1. On the second Tank Controller, use the Export command for the EZO probe to get the calibration strings, and upload those strings to the cloud server (oap.cs.wallawalla.edu, for example). This is the "copy" step.
2. On the first Tank Controller, download the calibration strings from the cloud server and use the Import command for the EZO probe to set up that calibration. This is the "paste" step.

jgfoster, 2024-01-09:
In fact, one would not need to put it on a server. A web client could simply get or put the current calibration data (120 hexadecimal digits). The web client could have a multi-line text entry field (10 lines) and a GET/PUT pair of buttons.

je-foster, 2024-01-09:
That should work nicely, but it would make the web client required for the feature.

je-foster, 2024-01-09:
I see, it avoids requiring the server, though. Probably worth it.